ArduPilot/ardupilot_wiki #947: Incompatible Align M480L ESCs & Motors - Pixhawk

223Wylde, 2017-09-25:
I added a Pixhawk to an Align M480L a while back, and eventually discovered both the Align ESCs and motors were incompatible with the Pixhawk.   The Align ESCs and Motors that come with the M480L kit are designed for Align's flight controller, and you cannot program them, so as far as I know, the best solution is to just replace both with known compatible components as I did, then its an excellent quad rotor.    Before discovering this problem, I had one serious crash and several minor ones being if you don't replace the ESCs and Motors, it will fly, but the motors will simply not respond to high or rapid throttle requests from the flight controller and it will just flip over or tumble, I believe due to frequency incompatibility issues.

ChristopherOlson, 2017-09-26:
I reviewed the logs on this issue posted to the discuss forum. This was with Copter 3.3.3 running untuned with the default PID's. The log shows before the crash that the aircraft was in a consistent ~20 degree nose-down attitude, ~8 degrees left roll, with RCOUT.1 operating consistently at ~90% throttle @ ~7m/s ground speed. Indicating the aircraft was fighting wind. At ~555 seconds in the log the aircraft hit a gust of wind, ran out of throttle overhead due to being mis-tuned, unpowered or under-propped for the conditions.

At no time do I see out-of-range output by the ArduPilot system on channels 1 thru 4. At the point of yaw deviation (40 degrees from desired) the software throttled the motors in an attempt to correct yaw, lost thrust on the roll axis due to being maxed out, the aircraft flipped and crashed. Power consumption peaked at 1,375 watts during the flight controller's attempts to regain control, which corresponds with calculating the approximate power based on throttle % to all four motors at that point. Indicating all motors were working properly.

I do not believe this to be a problem with the Pixhawk hardware or ArduPilot software having a compatibility issue with the downstream hardware. Top speed of this aircraft, according to the spec's provided by Align is 15 m/s. It appears the 370Kv motors used on this aircraft were simply operating too close to the top end of their rpm range to be effective at stabilizing the aircraft in the wind at the point of loss of control (and possibly not having the PID's tuned).

[image]

auturgy, 2017-10-06:
Thanks Chris, for the log analysis, and letting us know!
@223Wylde I've looked at the log and agree with @ChristopherOlson that this appears to be a control/thrust issue, not a hardware compatibility issue.  Can we close this?

223Wylde, 2017-10-06:
Sure, go ahead and close it.

Thanks,
Jeff

closed 2017-10-06 by OXINARF