ArduPilot/ardupilot_wiki #948: Building ArduPilot for Pixhawk/2/PX4 on Linux with Make

gpetty, 2017-09-25:
[Building ArduPilot for Pixhawk/2/PX4 on Linux with Make](http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html) 

Under Ubuntu 14.04, I found that  'make px4-v3' generates the error, 'No module named builtins'.   On my system, this was solved by executing the command 'pip install future --upgrade'

make then ran a while before generating another error 'No module name em', and prompting me to execute

sudo apt-get install python-empy
sudo pip install catkin_pkg

The first command gave the message that the latest version was already installed.  The second completed, but then when re-running 'make', the same messages were generated again -- i.e., no change. 

In short, I'm stuck.

magicrub, 2017-09-25:
just to confirm, you ran this command:
`Tools/scripts/install-prereqs-ubuntu.sh -y
`
 and it did not fix all dependencies?

gpetty, 2017-09-25:
Correct.

Appended at the bottom is my command history (bash) with some extraneous stuff removed.

The 'make' command error was the first indication of trouble.

  627  sudo apt-get -qq -y install git
  631  git clone https://github.com/ArduPilot/ardupilot.git
  632  cd ardupilot
  634  git submodule update --init --recursive
  647  Tools/scripts/install-prereqs-ubuntu.sh  -y
  648  cd ArduPlane
  652  make px4-v3
  653  ~/.profile
  659  source ~/.profile  # <-  Note, I also opened a fresh terminal window
  660  make px4-v3
  661  pip install future --upgrade
  662  make px4-v3
  663  sudo apt-get install python-empy
  664  sudo pip install catkin_pkg
  665  make px4-v3
  666  sudo apt-get install python-empy
  667  sudo pip install catkin_pkg
  668  sudo pip install catkin_pkg --upgrade
  669  make px4-v3

OXINARF, 2017-09-27:
Clearly your run of the _prereqs_ script wasn't successful for some reason. Without having the error from it, there is no way to tell what went wrong.

https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/install-prereqs-ubuntu.sh#L7-L10 - you can clearly see *future* and *python-empy* as packages to be installed. *catkin_pkg* isn't needed.

Have you tried to run `make px4-clean` and then try to build again?

gpetty, 2017-09-27:
I ran 'make px4-clean' and then re-ran 'make px4-v3'.   Here's a link to the captured output: https://www.dropbox.com/s/hnwdxioap0j3tgo/make.log?dl=0

As before, if I try to install empy or catkin_pkg as directed (see above output), I'm told I already have the latest versions.

OXINARF, 2017-09-27:
Have you tried to reinstall *python-empy*? Quite strange you have it, but the module isn't found by Python. In case you can't get it to work you can try with pip: `sudo -H pip install empy`

gpetty, 2017-09-27:
I think I found the problem:  I unknowingly had two conflicting python distributions installed.  When I used the 'pip' or 'python' command at the command line, it was looking in /usr/lib/python2.7/.  The makefile, on the other hand, was apparently getting python from /home/parallels/anaconda/bin.  Offhand, I'm not sure why they weren't both using the same path.

I executed 'sudo -H /home/parallels/anaconda/bin/pip install empy', and that seems to have cleared the logjam -- 'make' then completed successfully. 

Since I have often had machines with multiple distributions of python on them (for reasons I didn't control), I wonder if there's a way to diagnose and/or resolve potential path conflicts as part of the build process.

magicrub, 2017-09-27:
great, glad its working! Closing...

closed 2017-09-27 by magicrub

shivam5594, 2018-03-17:
I had all the required packages installed still it was failing. Then I tried make distclean. It worked after that. and another issue with the code is it gives error like :

`[158/158] Linking CXX executable nuttx_px4io-v2_default.elf
[375/945] Building CXX object src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V2 -DMODULE_NAME=\"bmp280\" -DPX4_MAIN=bmp280_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -Isrc/modules/systemlib/param -Iexternal/Install/include -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Wframe-larger-than=1024 -MMD -MT src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj -MF src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj.d -o src/drivers/barometer/bmp280/CMakeFiles/drivers__bmp280.dir/bmp280_spi.cpp.obj -c ../../src/drivers/barometer/bmp280/bmp280_spi.cpp
../../src/drivers/barometer/bmp280/bmp280_spi.cpp: In constructor 'BMP280_SPI::BMP280_SPI(uint8_t, uint32_t, bool)':
../../src/drivers/barometer/bmp280/bmp280_spi.cpp:90:67: error: declaration of 'external' shadows a member of 'this' [-Werror=shadow]
 BMP280_SPI::BMP280_SPI(uint8_t bus, uint32_t device, bool external) :
                                                                   ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[375/945] Generating nuttx/libc/libc.a
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1
`

simply changed the variable name and it worked.

magicrub, 2018-03-17:
What did you change? Can you make a pull request for it?

shivam5594, 2018-03-17:
Just made the pull request.