ArduPilot/ardupilot_wiki #950: SITL Simulator (Software in the Loop)

yanshuang16, 2017-09-26:
[SITL Simulator (Software in the Loop)](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
How the SITL simulate the powers' consumption? Where is the code?

hamishwillee, 2017-09-26:
I don't know - normally I run out of power really quickly :-)

Anyway, when this is known, we might want to update http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#

OXINARF, 2017-09-27:
It's a pretty simple simulation, doesn't try to be very close to reality: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_SITL/SITL_State.cpp#L408-L429

@yanshuang16 By the way, this issues list is for problems in the wiki or requesting that something is documented. For questions, it's best to post in the forum (http://discuss.ardupilot.org); if the question is development related you can also ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

closed 2017-09-27 by OXINARF