ArduPilot/ardupilot_wiki #958: Setting up SITL on Linux

ekkrider, 2017-09-30:
I'm trying to set up SITL on linux (I'm using Ubuntu 14.04) by following the steps on http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html

However, I got an error when I ran _sim_vehicle.py -w_ in my ardupilot/ArduCopter directory: AttributeError: 'sitl' object has no attribute 'with_uavcan'

Below is what I got after running _sim_vehicle.py -w_ in terminal:

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/ekkrider/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/ekkrider/ardupilot 
Setting out to                           : /home/ekkrider/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for need to link with librt     : not necessary 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.6 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (0.796s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/ekkrider/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/ekkrider/ardupilot/build/sitl'
Traceback (most recent call last):
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Scripting.py", line 266, in run_commands
    ctx = run_command(cmd_name)
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Scripting.py", line 250, in run_command
    ctx.execute()
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Scripting.py", line 616, in execute
    return execute_method(self)
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Build.py", line 255, in execute
    self.execute_build()
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Build.py", line 268, in execute_build
    self.recurse([self.run_dir])
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/ekkrider/ardupilot/wscript", line 361, in build
    _build_recursion(bld)
  File "/home/ekkrider/ardupilot/wscript", line 322, in _build_recursion
    bld.recurse(d)
  File "/home/ekkrider/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/ekkrider/ardupilot/ardupilot/wscript", line 383, in build
    if bld.get_board().with_uavcan:
AttributeError: 'sitl' object has no attribute 'with_uavcan'
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks

OXINARF, 2017-10-02:
This issues list is for problems in the wiki. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

It looks like your files aren't all updated. Check if you have this line in your files: 
https://github.com/ArduPilot/ardupilot/blob/master/Tools/ardupilotwaf/boards.py#L33

If you don't, then you need to reset your branch to fix your files. If you do, I suggest trying `sim_vehicle.py` with the *--clean* option.

closed 2017-10-02 by OXINARF

ekkrider, 2017-10-02:
I'm sorry for posting in the wrong place! I'll post there next time.

Thank you very much for your time and assistance!