ArduPilot/ardupilot_wiki #964: Broken links: Supported AutoPilot Controller Boards

DawoodTaii, 2017-10-07:
[Supported AutoPilot Controller Boards](http://ardupilot.org/dev/docs/supported-autopilot-controller-boards.html) -

A lot of links on this page is broken specially all links under 
http://firmware.ardupilot.org/downloads/wiki/pdf_guides/

I couldn't find alternative links for them. I think maybe pdf_guides directory is dropped by mistake or server error, may be it can be recovered?

DawoodTaii, 2017-10-07:
Also all links under auav.com is broken, the whole old website is taken down.

auturgy, 2017-10-15:
Thanks for letting us know!  Could you have a go at fixing these, and putting a comment linking a pull request in here?  You'll find an "edit on github" button near the top right had side of the page.

hamishwillee, 2017-10-15:
@auturgy The PDF guides were hosted on the firmware server. When uploaded those links work, so has the firmware server been moved. @jpkh should know.

auturgy, 2017-10-16:
@hamishwillee the links to the pdf guides themselves are broken, as are almost every hyperlink within the pdf's.  Would be great if @DawoodTaii could have a go at fixing!

OXINARF, 2017-10-17:
The correct link would be http://download.ardupilot.org/downloads/wiki/pdf_guides/ but even there a lot of PDFs are missing.

Like many pages in the dev section, this one also needs to be redone.

hamishwillee, 2017-10-17:
@OXINARF @auturgy Just FYI, it is possible to use sphinx to build with a link checker option. Might be worth doing that soon and doing a global attempt to chase broken links. The last time this was done was early last year.

Hwurzburg, 2019-11-06:
pdfs have been removed...closing

closed 2019-11-06 by Hwurzburg