ArduPilot/ardupilot_wiki #974: License (GPLv3)

auturgy, 2017-10-12:
[License (GPLv3)](http://ardupilot.org/dev/docs/license-gplv3.html) - Fix contact email (note to remind me).

magicrub, 2017-10-12:
Nice catch, we'll find a new contact and update it. Thank you!

auturgy, 2017-10-14:
@CraigElder suggestions? Should this link to the contacts page, or is there a particular email address we can use?
I'd prefer email (as would people/companies wanting to contact us in this regard), but you'd probably be the one at the end of it, so your call I think.

CraigElder, 2017-10-17:
I have been meaning to fix this too

[image]

CraigElder, 2017-10-17:
Updated https://github.com/ArduPilot/ardupilot_wiki/commit/9f2dd346cd8b7c47c2a23afb72c084ca97f29d05

closed 2017-10-17 by CraigElder