ArduPilot/ardupilot_wiki #982: Hyperlinks need checking throughout the wiki

auturgy, 2017-10-19:
ref https://github.com/ArduPilot/ardupilot_wiki/issues/964
This issue is opened to flag the requirement to check for broken links throughout the wiki.
Internal links can be checked via sphinx, however many "broken" links are due to external re-directs, so this task will require manual effort to complete.  
Partners/vendors - please check links to your products/sites and submit PR's where appropriate.
All - please add pages with broken or incorrect links into a comment below.

auturgy, 2017-10-19:
http://ardupilot.org/plane/docs/common-pixracer-overview.html#common-pixracer-overview
Pixracer hyperlink goes to auav.co redirect page.  Needs to point to the mRo product

Naterater, 2018-01-24:
This page needs a lot of help.
http://ardupilot.org/plane/docs/ecalc-motor-and-prop-efficiency-guide.html

Hwurzburg, 2019-11-06:
pages mentioned fixed or removed....

brunoolivieri, 2019-11-13:
I found some GitHub broken links. I guess that was linked for a current version in past time (this phrase makes sense?)

| File | Link |
|------|------|
|.\common\source\docs\common-ppm-encoder-apm2x-atmega32u2.rst    |    https://github.com/ArduPilot/ardupilot/blob/master/Tools/ArduPPM/Libraries/PPM_Encoder.h
|.\common\source\docs\common-qualcomm-snapdragon-flight-kit.rst    |    https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_Linux/qflight
|.\common\source\docs\common-rangefinder-objectavoidance.rst    |    https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.h#L24
|.\common\source\docs\common-zed.rst    |    https://github.com/yankailab/OpenKAI/blob/master/kiss/apmCopter.kiss
|.\dev\source\docs\apmcopter-programming-libraries.rst    |    https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Buffer
|.\dev\source\docs\building-for-erle-brain.rst    |    https://github.com/ArduPilot/ardupilot/blob/master/mk/targets.mk#L75
|.\dev\source\docs\editing-the-code-with-eclipse.rst    |    https://github.com/ArduPilot/ardupilot/blob/master/mk/px4_targets.mk

hamishwillee, 2019-11-13:
This is built-in to sphinx:
```
make linkcheck
```
See https://www.writethedocs.org/guide/tools/testing/

Hwurzburg, 2019-11-14:
if we used that now, it would never build....we currently have ~100 dead links on the website and subsections...(down from 250 when I first looked over a month ago...so thats good...we've been fixing them as we work on the wiki)....

a lot of them generated from archived pages that will ALWAYS generate bad links...

deadlinkchecker.com if run on Http://ardupilot.org will show the list....like on the ArduPlane index page, the target directive is wrong for the first image...it will generate a 404 if you click on the image (why we have hyperlinked images to themselves mystifies me, but that is another topic itself)...just havent gotten around to fixing that yet...

Bruno is tackling some I see...

hamishwillee, 2019-11-14:
Of course. I used to run this locally on occasion and post results to an issue so that others could fix. The point is "no need to build custom tools" to find the problem cases.

With respect to archived pages, it is worth considering whether it is worth introducing versioned builds. For example, much of the archived material corresponds to APM2.7 FCs. If you drew a line in the sand most of that could be in "old version", allowing new version to be created as a clean slate. Something to think about anyway.