ArduPilot/ardupilot_wiki #1001: Gazebo with SITL - simplify instructions

Katawann, 2017-11-08:
[Using Gazebo Simulator with SITL](http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html) - update the tutorial

I installed Gazebo Simulator to work with SITL and it works perfectly. However, I think the tutorial is a little bit hard to understand (a lot of possibilty to install but few of them are really fastidious and did not work for me). Maybe an update would be nice. I share here the steps I did to connect SITL to Gazebo, which could be useful for others (the installation has only been tested with Ubuntu 16.04 LTS).

**Install Gazebo**
Install Gazebo as proposed on the [website](http://gazebosim.org/tutorials?tut=install_ubuntu)
1.	Install
`curl -ssL http://get.gazebosim.org | sh`
2.	Run
`gazebo`

**ArduPilot SITL plugin**
To communicate with SITL, it is necessary to add a plugin taken from [here](https://github.com/swiftgust/ardupilot_gazebo):

1. Add the plugins for ArduPilot SITL
```
git clone https://github.com/SwiftGust/ardupilot_gazebo
cd ardupilot_gazebo
mkdir build
cd build
cmake ..
make -j4
sudo make install
```

2. Set Path of Gazebo Models
```
echo 'export GAZEBO_MODEL_PATH=~/ardupilot_gazebo/gazebo_models' >> ~/.bashrc
source ~/.bashrc
```
3. Copy Demo Worlds to Gazebo

`sudo cp -a ~/ardupilot_gazebo/gazebo_worlds/. ~/<your_working_space>/worlds`

And that's it !

**Run Gazebo and SITL**
On 1st Terminal (Launch Ardupilot SITL)
`sim_vehicle.py -v ArduCopter -f gazebo-iris --mav10 --console -I0`
or only:
`sim_vehicle.py -j4 -f Gazebo`
On 2nd Terminal (Launch Gazebo with demo 3DR Iris model)
`gazebo --verbose worlds/iris_irlock_demo.world`

If you have better solution I would be glad to hear it! I still have some difficulties to communicate between two machines (a powerful machine to run Gazebo and my laptop for SITL/Mission Planner) so if anyone have an idea how to do it, please share :) .

Katawann, 2017-11-08:
Ok I just found a really good tutorial for installing step by step SITL, plugin and Gazebo [here](https://github.com/AS4SR/general_info/wiki/ArduPilot:-Instructions-to-set-up-and-run-an-autopilot-using-SITL-and-Gazebo-simulator)

Katawann, 2017-11-08:
Last comment for my monologue: ) I finally understood how to work with Mission Planner and Gazebo which is more convenient to test some flights in my opinion.

**Use Mission Planner for control**
To control the simulation from Mission Planner (Windows only) it is necessary to forward the packet from MAVProxy to Mission Planner. It can be done in two ways:
When running the script sim_vehicle.py, add the following parameters:
`--mavproxy-args “—out=udp:<ip_target>:14550”`

Or directly from the consol with :
`add output <ip_target>:14550`

**Warning**: do not forget to open the port on your target (running Mission Planner) to be able to control it)

khancyr, 2017-11-08:
Yep I agree that it is simplier. I need to finish the plugin correctly ... If you got it working directly, it means that the new release of gazebo fix the path issue. 
Thanks for sharing your experience ! I will update the wiki !

umer936, 2017-11-18:
+1. I would really, really like it if the wiki for Gazebo simulation was updated and clarified. Some of the instructions seem to contradict each other and have left me very confused. Thanks in advance!

anbello, 2018-02-24:
I tried to follows this instruction (and those on https://github.com/swiftgust/ardupilot_gazebo) but i otain the following error:

```
andrea@galileo:~$ gazebo --verbose worlds/iris_irlock_demo.world
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.10.16
[Msg] Publicized address: 192.168.10.16
Warning [parser.cc:536] XML Attribute[xmlns:xacro] in element[sdf] not defined in SDF, ignoring.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Err] [Joint.cc:260] EXCEPTION: Couldn't Find Parent Link[iris::iris_demo::gimbal_small_2d::tilt_link]

[Err] [Model.cc:208] LoadJoint Failed
[Err] [InsertModelWidget.cc:298] Missing model.config for model "/home/andrea/ardupilot_gazebo/gazebo_models/.hg"
[Dbg] [ArduPilotPlugin.cc:1128] [iris] ArduPilot controller online detected.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Err] [Scene.cc:2927] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Err] [Scene.cc:2927] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.

```

with the following output in Gazebo:

https://github.com/anbello/Hello-World/blob/master/Screenshot%20from%202018-02-24%2018-23-32.png

anbello, 2018-02-25:
Thamks @Katawann 
I already added GAZEBO_MODEL_PATH to my .bashrc but I have the error posted above.
I would like to use the camera for experimenting with aruco marker detect with ROS.

anbello, 2018-02-25:
@Katawann  the problem was in the gazebo models coming with https://github.com/SwiftGust/ardupilot_gazebo if I remove the GAZEBO_MODEL_PATH from my .bashrc then gazebo says that will get models from http://models.gazebosim.org/ and seems all OK.
It remains only the error relative to light (sun) and the warning relative to conversion of sensor (imu), do you know what they means?

Thanks

```
[Msg] Waiting for master.
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.10.16
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.10.16
[Wrn] [ModelDatabase.cc:339] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[imu] not suppported.
[Err] [Scene.cc:2927] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
[Err] [Scene.cc:2927] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
[Dbg] [ArduCopterPlugin.cc:722] ArduCopter controller online detected.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[Err] [Scene.cc:2927] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
[Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
```

Katawann, 2018-02-25:
@anbello 
So back to your problem with the gimbal, I actually had the same problem but didn't remember I fixed it... I create a [commit ](https://github.com/Katawann/ardupilot_gazebo/commit/6e7a48ef6a52cc13240b65fbbf0a4055865c5f89) with the fixe so you can have a look. You may have some problems with your solution to remove the model path of your bash if you want to add or modify some models later . Mainly the gimbal should not be include in the world file but in the iris model in my opinion. If it fixes your solution for the gimbal, feel free to do a pull request to SwiftGust project. I dont have that much time to do it yet.

For the warning problem you can ignore it. I also have it but it doesn't affect my simulation in any case

And for your [sun] error, I won't be able to help you because I never had it. But I think it is more related to Gazebo than the plugin. Does it work when you run other worlds than the irlock_demo ?

anbello, 2018-02-26:
@Katawann 
Now i can't test your commit I will do asap.
Anyway when you remove the model path gazebo download the necessary models in ~/.gazebo/models and the successive time it needs them it uses the local copy, so I think you can also modify them.

For your last question I tested iris_irlock_demo.world and iris_ardupilot.world and they works.

anbello, 2018-02-27:
@Katawann your commit works for me, now I have two solutions: using models downloaded from gazebo site or using your commit, thanks.

Katawann, 2018-02-27:
thanks for the test and feedback ! I'll do a pull request asap

umer936, 2018-10-07:
ROS, ArduPilot, Gazebo installation instructions still aren't great. 
So my team has since written seperate instructions for our use-case, but our documentation may be helpful in updating this one. https://github.com/Texas-Aerial-Robotics/Controls-ROS

khancyr, 2019-01-24:
Instructions have been updated and cleaned. please open other issues if something is wrong

closed 2019-01-24 by khancyr

Rezenders, 2019-06-26:
> I still have some difficulties to communicate between two machines (a powerful machine to run Gazebo and my laptop for SITL/Mission Planner) so if anyone have an idea how to do it, please share :) .

Did you find a solution for this?

Sorry for necrobumping