ArduPilot/ardupilot_wiki #1015: Compliation of Arduplane code in Ubuntu 16.04 64 bit

dinesh286, 2017-11-23:
I followed the procedure in the link http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html, but after installation I complied in using "make px4".  I got the following error

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

/home/dinesh/Ardupilot/ardupilot/modules/PX4Firmware/Makefile.make:262: recipe for target 'generateuorbtopicheaders' failed
make[2]: *** [generateuorbtopicheaders] Error 1
make[2]: Leaving directory '/home/dinesh/Ardupilot/ardupilot/modules/PX4Firmware'
../mk/px4_targets.mk:191: recipe for target 'px4-io-v1' failed
make[1]: *** [px4-io-v1] Error 2
make[1]: Leaving directory '/home/dinesh/Ardupilot/ardupilot/ArduPlane'
../mk/px4_targets.mk:143: recipe for target 'px4' failed
make: *** [px4] Error 2
  

After installing empy also i m getting the same error, please tell be how to go about

magicrub, 2017-11-23:
The Ubuntu script already includes [python-empy](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/install-prereqs-ubuntu.sh#L10) , have you run the script? It's listed on the page you linked

Why is catkin_pkg needed?

dinesh286, 2017-11-23:
I installed both empy and catkin_pkg, but still I have the same problem

magicrub, 2017-11-23:
Did you run the ubuntu script?

dinesh286, 2017-11-24:
S I did.  Followed the quick guide setup. Loged out and the I try building

dinesh286, 2017-11-24:
The issue is solved. i could compile aft following the instructions in the link https://github.com/ArduPilot/ardupilot_wiki/issues/948.
I have another question.  the advanced settings mentioned in the page http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html is required for HK Pixhawk2?
Is there any .sh file for the installation and settings?

magicrub, 2017-11-24:
So, the problem was you had multiple installations of Python? It was a cross-VM-weirdness issue?

magicrub, 2017-11-24:
>I have another question. the advanced settings mentioned in the page http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html is required for HK Pixhawk2

I'm not sure why that "advanced" message is there. It should all be handled in the same script. Sounds like you're compiling successfully so I'm closing.

closed 2017-11-24 by magicrub