ArduPilot/ardupilot_wiki #1016: No map shows up

Corleno, 2017-11-25:
#Hi ardupilot contributors,

I have a question about the map of simulator. I try to run sim_vehicle.py -j4 --map --console. But no map shows up. In the map part, it shows unavailable for every pixel. And for terminal, it shows as followed. Could you please tell me how to settle this issue?

### _Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (5.311s)
SIM_VEHICLE: Using defaults from (/home/rui/Documents/Ardupilot/ardupilot/Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/rui/Documents/Ardupilot/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/rui/Documents/Ardupilot/ardupilot/build/sitl/bin/arducopter" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "--model" "+" "--speedup" "1" "--defaults" "/home/rui/Documents/Ardupilot/ardupilot/Tools/autotest/default_params/copter.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--map" "--console"
RiTW: Starting ArduCopter : /home/rui/Documents/Ardupilot/ardupilot/build/sitl/bin/arducopter -S -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults /home/rui/Documents/Ardupilot/ardupilot/Tools/autotest/default_params/copter.parm
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
STABILIZE> Received 886 parameters
Saved 886 parameters to mav.parm
/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/mp_util.py:154: wxPyDeprecationWarning: Accessing deprecated property. 
  a = getattr(object, v)

STABILIZE>_

auturgy, 2017-11-25:
This was identified as a map source issue, complicated by the Unity desktop putting menus in strange places.
Resolved.

closed 2017-11-25 by auturgy

tokyokuma, 2017-12-20:
I am facing the same problem.
I think that it is probably solved in the master branch, but I use the branch of Rover-3.2.
In my case, where is the map source and where do I move it?

rmackay9, 2017-12-20:
@tokyokuma, it's probably the map service being used.  Assuming you are using SITL/MAVProxy on Ubuntu, click on the map itself, then hover over the "Console" on the top left of the screen and a "View" menu should appear.  Select Service and then OviHybrid.  At least that works for me and I'm in Japan.. which I assume is where you are as well.

tokyokuma, 2017-12-21:
Sorry, I was short of confirmation. I got a map! Thank you!