ArduPilot/ardupilot_wiki #1017: Using Gazebo Simulator with SITL

patrickelectric, 2017-11-25:
[Using Gazebo Simulator with SITL](http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html) - The name now is gazebo-zephyr and no iris.

patrickelectric, 2017-11-25:
```
{
  'quadplane-tilttri': {
    'make_target': 'sitl',
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/quadplane-tilttri.parm'
  },
  'quadplane-tilttrivec': {
    'make_target': 'sitl',
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/quadplane-tilttrivec.parm'
  },
  'quadplane-tilthvec': {
    'make_target': 'sitl',
    'waf_target': 'bin/arduplane',
    'default_params_filename': [
      'default_params/plane.parm',
      'default_params/quadplane-tilthvec.parm'
    ]
  },
  'quadplane-tri': {
    'make_target': 'sitl',
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/quadplane-tri.parm'
  },
  'quadplane-cl84': {
    'make_target': 'sitl',
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/quadplane-cl84.parm'
  },
  'quadplane': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/quadplane.parm'
  },
  'firefly': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/firefly.parm'
  },
  'plane-elevon': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': [
      'default_params/plane.parm',
      'default_params/plane-elevons.parm'
    ]
  },
  'plane-vtail': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': [
      'default_params/plane.parm',
      'default_params/plane-vtail.parm'
    ]
  },
  'plane-tailsitter': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/plane-tailsitter.parm'
  },
  'plane': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/plane.parm'
  },
  'plane-dspoilers': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': [
      'default_params/plane.parm',
      'default_params/plane-dspoilers.parm'
    ]
  },
  'gazebo-zephyr': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/gazebo-zephyr.parm'
  },
  'last_letter': {
    'waf_target': 'bin/arduplane'
  },
  'CRRCSim': {
    'waf_target': 'bin/arduplane'
  },
  'jsbsim': {
    'waf_target': 'bin/arduplane',
    'default_params_filename': 'default_params/plane-jsbsim.parm'
  },
  'calibration': {
    'extra_mavlink_cmds': 'module load sitl_calibration;'
  }
}cmd_opts)
```

patrickelectric, 2017-11-25:
and in gazebo: `gazebo --verbose worlds/zephyr_demo.world`

OXINARF, 2017-11-25:
I think it has been clarified in Gitter. `gazebo-iris` still exists and is a Copter, `gazebo-zephyr` is a Plane. There is also `gazebo-rover` although I'm not sure which Gazebo model that should use.

closed 2017-11-25 by OXINARF