ArduPilot/ardupilot_wiki #1024: Selecting plane conf in arduplane

dinesh286, 2017-11-30:
I could compile arduplane code in ubuntu and upload using mission planner.  I want to know how to select a plane conf in arduplane code.  Arduplane supports different airplane conf, like fixed wing with conventional tail conf, v tail conf, flying wing.  I want to choose a fixed wing with conventional tail conf (multiplex easy star, as shown in QGC)

magicrub, 2017-11-30:
Ardupilot.org -> Docs -> Plane -> First Time Setup -> Configuration -> [4 channel plane](http://ardupilot.org/plane/docs/guide-four-channel-plane.html)

closed 2017-11-30 by magicrub

dinesh286, 2017-12-01:
can i set the values of SERVOn_FUNCTION using mission planner.  Is it possible to set SERVOn_FUNCTION in the code directly, if so plz tell me where to set

magicrub, 2017-12-01:
Config -> Planner -> Layout -> advanced

dinesh286, 2017-12-01:
plz mention the site to search these menus you hve given

magicrub, 2017-12-01:
@meee1 These params are now required for setting up a plane in v3.8. instead of pushing everyone into Full-params to configure it, can we have a page where it shows common aircraft types and changes these servo functions for you?