ArduPilot/ardupilot_wiki #1030: Autonomous Landing for a Fixed wing RC plane (like Skywalker)

dinesh286, 2017-12-04:
I want to make a fixed wing plane(like skywalker) to do a autonomous landing using own algorithm.  I recently downloaded the arduplane code and compiled in ubuntu 16.04.  I want to know how and where to edit the arduplane code to put my own algorithm.   Is there any other better option to do this .

magicrub, 2017-12-04:
Looks like you've posted a help question on the wiki Todo list. Closing. Please post help questions on the forum so the proper people can help you out! Http://discuss.ardupilot.org

magicrub, 2017-12-04:
Please check this page before you make any new posts
http://ardupilot.org/plane/docs/automatic-landing.html

closed 2017-12-04 by magicrub

peterbarker, 2017-12-05:
On Mon, 4 Dec 2017, dinesh286 wrote:

> I want to make a fixed wing plane(like skywalker) to do a autonomous
> landing using own algorithm. I recently downloaded the arduplane code and
> compiled in ubuntu 16.04. I want to know how and where to edit the
> arduplane code to put my own algorithm. Is there any other better option to
> do this .

All of what Tom said, but additionally - use SITL - and perhaps autotest - 
to develop your new algorithm.  That's also on the Wiki :-)

dinesh286, 2017-12-05:
can u send me the link of wiki

magicrub, 2017-12-05:
The link was already posted above. This is closed. Please use discuss.ArduPilot.org for help questions.