ArduPilot/ardupilot_wiki #1062: Copter Object Avoidance == please add Control-F option to show RADAR

patrickpoirier51, 2018-01-17:
[Copter Object Avoidance](http://ardupilot.org/dev/docs/code-overview-object-avoidance.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Lots of questions on the ''disapearing'' of the RADAR screen.
There is no information about CTRL-F to  invoque option screen and select the proximity display - RADAR

rmackay9, 2018-01-17:
yes I totally agree on this one.  I just found out yesterday that the ctrl-f is required.  I'll add it to the wiki.

rmackay9, 2018-01-19:
I've added some extra info on this at the bottom (changes should appear about 10min from the time I did this post).

closed 2018-01-19 by rmackay9