ArduPilot/ardupilot_wiki #1064: document Servo input/output pass through feature

rmackay9, 2018-01-20:
In Copter-3.5.4 (and higher) we support passing through the input received on any channel to any output channel.  This is done by setting SERVOx_FUNCTION to a number from 51 to 66.

For example, setting SERVO9_FUNCTION = 58 would make a Pixhawk's "AUX OUT 1" channel (aka output channel 9) output the value that's being received on input channel 8

Ryanf55, 2018-03-23:
Could the documentation be more clear on how the endpoints, trim, and reversal works for SERVOn_FUNCTION when set to values of 51 - 66 ( RC_INn). Basically, does the RC in control trim, or is it overwritten by Servon_Function? What about the endpoints?

Hwurzburg, 2019-10-11:
addressed by #2068

closed 2019-11-10 by Hwurzburg