ArduPilot/ardupilot_wiki #1067: Explain ArduPilot firmware architecture with diagrams?

k4kfh, 2018-01-23:
As a new user, I'm a little overwhelmed by all the different pieces of a functioning Pixhawk running Ardupilot. Something to explain the different kinds of files (``.px4``, ``.apj``, and any others) and the different pieces of software (such as NuttX) that go into making all of this work would be a huge help.

If someone can explain that to me I will be happy to create some nice looking diagrams layer-cake style, as I think that will make more sense to newbies like myself.

k4kfh, 2018-01-24:
Whoops...apparently I am just blind. Those seem to answer all my questions. Thank you! Sorry to waste time.

closed 2018-01-24 by k4kfh

rmackay9, 2018-01-24:
No troubles, thanks for your interest in ardupilot!