ArduPilot/ardupilot_wiki #1069: Rover: add wiring diagram for BrushedWithRelay motor type

rmackay9, 2018-01-26:
Our Motors & Servo wiki page describes how to connect the flight controller to motor driver's using "Normal" pwm input (i.e. in the range 1ms ~ 2ms).  We should add wiring diagrams and instructions on how BrushedWithRelay and perhaps BrushedBiPolar motor types can be used.  Ideally we should include links to motor drivers that use these inputs.

These pages should be added to:
http://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html
http://ardupilot.org/rover/docs/rover-motor-and-servo-configuration.html#rover-motor-and-servo-configuration

Chasvortex, 2018-07-27:
This would be mega helpful - if anyone knows how it works please help!

FabehaRaheel, 2023-04-04:
Hi,
Please do update the wiring diagrams and hardware setup instructions for brushed dc motors.  
Thanks!

yuri-rage, 2024-10-28:
This is still open and really needs an update. https://github.com/ArduPilot/ardupilot_wiki/issues/6351 would benefit as well from a diagram.

PR #6961 (2025-06-30): Brushed Motors: add L298N native support instructions
As of 4.6, AP supports inverted relay outputs, deprecating the previously linked script, which was an incomplete and slightly awkward implementation.

This should close #1069 and #6351. It also renders https://github.com/ArduPilot/ardupilot/pull/28488 unnecessary, though the PR script could be a means to more gracefully deprecate the previously linked script here.
closed 2025-06-30 by Hwurzburg