ArduPilot/ardupilot_wiki #1070: Rover: add list of known working motor drivers

rmackay9, 2018-01-26:
In the Assembly section of the rover wiki, perhaps on the [motor-and-servo connections page ](http://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html) we should add a list of known working motor drivers to allow the DIYers to pick a motor driver that we know works.

Hwurzburg, 2020-01-15:
anyone have a list? I will add....

rmackay9, 2020-01-15:
I know that these ones work:

- [RoboClaw 2x7a](https://www.pololu.com/product/3284) ([from this ArduRoller reference page](https://ardupilot.org/rover/docs/reference-frames-arduroller.html))
- [BlueRobotics Basic ESCs](https://bluerobotics.com/store/thrusters/speed-controllers/besc30-r3/) ([from the boogie board boat reference vehicle](https://ardupilot.org/rover/docs/reference-frames-boogieboard-boat.html))

closed 2020-02-03 by Hwurzburg