ArduPilot/ardupilot_wiki #1084: Compeating Compuses

thomannjf, 2018-02-08:
[](http://ardupilot.org/planner/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

thomannjf, 2018-02-08:
As soon as I connect my solo mission planner keep repeating “compeating compuses”

rmackay9, 2018-02-12:
This issues list is for tracking to-do items for the wiki.  Support should go into the forums or possibly you might find help on the Solo users or ArduPilot.org facebook pages.

http://ardupilot.org/copter/docs/common-contact-us.html

In case it helps, the error message is probably "inconsistent compasses" and the cause could be interference from metal objects (perhaps in the ground) or it could be the COMPASS_ORIENT parameter has been changed from 0.

closed 2018-02-12 by rmackay9