ArduPilot/ardupilot_wiki #1092: SITL on Linux Map "Unavailable"

mehmetyldz87, 2018-02-21:
I encountered such a problem with map "Unavailable" . How can I solve this problem? Regards Mehmet
[image]

rmackay9, 2018-02-22:
This is not really the place to ask support questions but in any case, the issue is the map provider. Go to the top right of the screen and a menu item should appear that allows setting the provider.  I find MicrosoftSatellite works.  Sadly it's not remembered so it needs to be done each time.  I hear from @peterbarker that there is a command line option to set the map provider but I haven't found it.

closed 2018-02-22 by rmackay9

mehmetyldz87, 2018-02-22:
Thanks @rmackay9  , I apologize for the inconvenience. I did not hear from this discuss . I opened a topic 

https://discuss.ardupilot.org/t/sitl-on-linux-map-unavailable/26088

rmackay9, 2018-02-22:
@mehmetyldz87, no troubles!