ArduPilot/ardupilot_wiki #1095: Plane Mode Table/Cheatsheet

Naterater, 2018-02-22:
I think a high-level graphical overview of the flight modes would help teach users some of the capabilities of Ardupilot.

Would anyone be able to publish a table like this in the plane flight modes wiki (http://ardupilot.org/plane/docs/flight-modes.html) or teach me how?

[image]

magicrub, 2018-02-22:
Wow, that's a GREEEEEAT chart! Love it! I'm still in full travel mode in Australia but when I get back in a few days I'll makes some time to do wiki stuff together.

tridge, 2018-02-22:
good chat, but needs some minor corrections. For example, it doesn't differentiate between FBWA and training, but they are very different modes
FBWA should be: roll and pitch follow stick input, up to set limits
Training should be: manual control up to roll and pitch limits
FBWB: like FBWA, but with automatic height and speed control
CRUISE: like FBWB, but with ground course tracking and terrain following
AUTOTUNE: like FBWA, but learns attitude tuning while flying
ACRO: rate controlled mode, no attitude limits

WickedShell, 2018-02-23:
Plane sorta requires GPS at all times I thought? (Just from a sensor fusion/estimation/arming perspective)

Naterater, 2018-03-02:
https://github.com/ArduPilot/ardupilot_wiki/pull/1106

magicrub, 2018-03-02:
Thanks for the PR @Naterater !

closed 2018-03-02 by magicrub