ArduPilot/ardupilot_wiki #1096: Terrain Following uploading terrain data only when GPS is locked

chris669, 2018-02-23:
[Terrain Following](http://ardupilot.org/copter/docs/terrain-following.html) -
Just want to improve this. I tried this option following the indications and posting in the forum, no answer https://discuss.ardupilot.org/t/follow-terrain-with-range-sensor-error-failsafe-terrain-1/25724/4
My little experience in this: 
You say in Wiki: 
Upload the mission to the vehicle and execute the mission as you normally would in AUTO
I've tried it many time without success.
[image]
ter_alt ter_pend ter_load were always on 0. So no terrain data loaded.
And then I've a GPS lock in my office, this is not often the case due to large walls... And I've seen that at this point, the terrain datas had been uploaded.  
So my contribution would be to improve this: Why the process of uploading the terrain data is not done in the same time the mission is uploaded??? A lot of people are planning the missions from home, not from the fly place location. Is it possible to do that? Thanks. Chris

SlipstreamFPV, 2018-03-01:
Good suggestion. I hope they can implement this. Would also be good to manually download and add .HGT files from the USGS to the SD card and have them work too.

Hwurzburg, 2020-01-15:
I have added a note in the terrain following page at the beginning, regarding the need for GPS lock in order to download terrain data (although it is mentioned in the text later in the page)

the ability to download even when the autopilot does not have GPS lock seems to be a good enhancement request...I suggest entering it as an issue in the Ardupilot code wiki...

closing this wiki issue...

closed 2020-01-15 by Hwurzburg