ArduPilot/ardupilot_wiki #1097: Terrain Following Fly away with Maxbotix I2C sensor

chris669, 2018-02-23:
[Terrain Following](http://ardupilot.org/copter/docs/common-terrain-following.html) - Just want to add an information about this. I did some fly tests with my Maxbotix I2C sensor attached and well configured. No problems when planning Relative Height flights, but with Terrain Follow option selected:
1. I take off
2. Launch the Auto mode
3. As this kind of lidar is unable to measure a distance of 2 meters outside (this info should be mentionned somewhere) and as it's configurated MIN MAX 20 700 cm as your documentation says, there is a problem: the drone couldn't measure the distance because based only on Lidar informations as rmackay9 told me in a post. So the drone fly away. 
4. The drone keep on flying away, reach the Failsafe altitude limit and even the RTL function is not working
5. Only way to get the drone back was to select stabilize FM.

When I selected Terrain Following WITHOUT Maxbotic activated RNGFND_TYPE = 0, I was able to do a mission Terrain.
So my point of view: it's now dangerous activating a lidar (inapropriate for outside, I aggree) and launching an Auto mission, this causes a Fly away and desactivate RTL.
My solution: if the Flight Controller receives inacurrate datas from the Lidar, switch to Baro or GPS or something like a mixed alt calculated altitude.

[image]

Thanks,
Chris

rmackay9, 2018-02-27:
Thanks for the report.  What you've stumbled upon is indeed an issue and here is the issue in the main ardupilot issues list: https://github.com/ArduPilot/ardupilot/issues/4697.

In this particular case, setting the RNGFND_MAX to a lower number like 200 (for 2m) should resolve the issue.

I'm going to close this issue 'cuz we've already got the other one.

auturgy, 2018-03-09:
Reopened, as this issue is specific to the wiki, the other to the underlying cause.

chris669, 2018-03-09:
thanks @auturgy

magicrub, 2018-04-27:
wiki is fixed so closed. Thank you @auturgy

closed 2018-04-27 by magicrub