ArduPilot/ardupilot_wiki #1099: Smart RTL is not listed in the Flight Modes section.

lordneeko, 2018-02-27:
Need to add the Smart RTL mode to the Flight Modes section for Rover and Copter

peterbarker, 2018-02-27:
On Tue, 27 Feb 2018, lordneeko wrote:

> Need to add the Smart RTL mode to the Flight Modes section for Rover and
> Copter

Thanks!  Bit of an oversight.

Do you think you might be able to add something?  You can edit the Wiki 
using the "Edit" button on the top right-hand-side of each page.  This 
creates a Pull Request which can then be approved...

lordneeko, 2018-02-27:
I would be glad to...but I don't know anything about it! That's why I went
looking for it. :)

lordneeko, 2018-02-27:
Suppose I could read the code and figure it out

peterbarker, 2018-02-27:
On Tue, 27 Feb 2018, lordneeko wrote:

> Suppose I could read the code and figure it out

Or fly it and find out for real :-)

If you google a bit you should find the GSoC stuff including videos.

Peter

closed 2018-03-05 by lordneeko