ArduPilot/ardupilot_wiki #1104: Should All copyright lines (currently 2016) in conf files and others be updated to 2018?

lordneeko, 2018-03-01:
https://github.com/ArduPilot/ardupilot_wiki/blob/452a9d835551091006e6cf81333a8ff239609fa1/ardupilot/source/conf.py#L63

hamishwillee, 2018-03-01:
Probably you'd change to the range: `u'2016 - 2018, ArduPilot Dev Team.'` 

Generally speaking you update copyright files when you make modifications - so initial copyright is the first date, and if you did some further changes you would change to add extra dates. However since this a global config and we don't individually update the sub files I suggest that it should be changed.

The other option is to create a plugin that is able to work out the github change dates and apply copyright on a per-page basis. IMO not worth the effort. 

Others might have  a different opinion. It is not uncommon for corporates to have a standard "all dates must reflect current year".

lordneeko, 2018-03-01:
You "could" also use date.now().year

hamishwillee, 2018-03-01:
Yes, but that would update on every rebuild for every file.

lordneeko, 2018-03-01:
Yes, not ideal.