ArduPilot/ardupilot_wiki #1135: Copter: add RPM sensor setup page

rmackay9, 2018-03-19:
Traditional helicopters often use RPM sensors to measure the main rotor speed. We should creat a wiki page which describes how to setup these sensors.

This was raised by @bzucak on Gitter as well.

Gone4Dirt, 2019-07-21:
I have recently been going through the trials and tribulations of getting my rpm sensor working so am now more familiar with what is going on.  I will take on writing the Wiki.  Not 100% sure on everything so I am going to out a post to the community on the discourse to call for some knowledge share.  I will then compile it all into the wiki.

rmackay9, 2019-07-22:
@Gone4Dirt, great!

Gone4Dirt, 2019-08-16:
I have submitted the PR for this page.  It is here:
https://github.com/ArduPilot/ardupilot_wiki/pull/1914

Hwurzburg, 2019-12-23:
pr merged...closing

closed 2019-12-23 by Hwurzburg