ArduPilot/ardupilot_wiki #1145: Setting up the waf Build Environment on Windows using Cygwin

evgenee, 2018-03-29:
Hi,
I am trying to build a px4-v2 target with waf
 and get an error:
[image]

I think it is an unresolved issue

Thank you

gmorph, 2018-03-29:
No your doing it wrong.  The build.md has all the instructions.
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

Thanks, Grant.

rmackay9, 2018-03-30:
I went over the instructions on the wiki a few weeks ago (ardupilot.org/dev/docs/building-setup-windows-cygwin.html).  The most common problem is the path is too long.  It's not an ardupilot issue so much as a general path limitation in cygwin.

Here's the note right at the end of the wiki page:

> The build may fail if the file path to some files is too long. If the build fails, please try cloning ArduPilot into a directory very high in the directory structure (i.e. ~/ardupilot).

billbonney, 2018-05-06:
It won’t go away unless windows changes. It’s a windows issue. The solution is to substitute a drive letter at the path of build ardupilot closer to the root level directory.

rmackay9, 2018-05-07:
I think I'll close this.

closed 2018-05-07 by rmackay9