ArduPilot/ardupilot_wiki #1159: SmartRTL

ghdmmbyd, 2018-04-04:
how to make  land  rate of SmartRTL mode  lower ?

rmackay9, 2018-04-05:
Questions should really go in the forums, perhaps in the new Copter-3.6 category.
the LAND_SPEED parameter will make it slower I think but only for the last few meters.

closed 2018-04-05 by rmackay9

ghdmmbyd, 2018-04-05:
Thanks!  I have tried it but it has no effect.

rmackay9, 2018-04-05:
@ghdmmbyd, OK.  WPNAV_SPEED_DN will probably make it descend more slowly but then it will always descend more slowly during AUTO, Guided, RTL, SmartRTL.

It'd possible we need to make a code modification.  Can you post a log in the [Copter-3.6 discussion](https://discuss.ardupilot.org/c/arducopter/copter-36) so we can investigate?

ghdmmbyd, 2018-04-05:
OK, I know, the log is in the company, after the holidays, test, and post the log.