ArduPilot/ardupilot_wiki #1161: Battery failsafe documentation

WickedShell, 2018-04-06:
All the battery failsafe docs for the upcoming releases (Copter 3.6, Plane 3.9) are now stale. There are now independent low and critical actions, priorities between low/critical failsafes, and the failsafes are configured per battery monitor. See https://github.com/ArduPilot/ardupilot/pull/7213 for more details.

Hwurzburg, 2020-02-25:
@WickedShell Are documents now up to date? if not, what needs to be done?thanks

WickedShell, 2020-02-25:
As long as it's actually discussing that there are two stages of battery fail safe available then I think it's fine. A quick glance says it's there, although a bit hidden. It's been in the last two releases and may be worth promoting more. But it's at least in the docs at all now, so I'll close this.

closed 2020-02-25 by WickedShell