ArduPilot/ardupilot_wiki #1183: RE: Hello

silver2row, 2018-04-09:
Hello,

Are people still working on this project? If so, there are some other projects that may entertain and teach some ideas relating to the BBB/and variations on this subject.

Seth

P.S. I found some other BBBlue references and found this page via ArduPilot's Wiki. Enjoy! Oh and I am building a drone/UAV w/ the BBBlue for now. If anyone is interested in pitching in, let me know. Maybe we can promote ideas together. Theory!

silver2row, 2018-04-09:
Hello Again,

Sorry, I was typing up ideas relating to this link: https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/building-for-beaglebone-black-on-linux.rst.

Seth

auturgy, 2018-04-09:
Have a look at https://github.com/mirkix/BBBMINI
And http://ardupilot.org/copter/docs/common-beagle-bone-blue.html
There are a number of users and developers actively using beaglebone based flight controllers, I’m sure they’re able to answer questions.

Welcome to ArduPilot, and good luck with your project!

James

auturgy, 2018-04-09:
Not a wiki issue, so closing

closed 2018-04-09 by auturgy