ArduPilot/ardupilot_wiki #1192: Plane failsafe actions are very incomplete

WickedShell, 2018-04-11:
http://ardupilot.org/plane/docs/apms-failsafe-function.html is missing a lot of short/long failsafe actions. The battery failsafe information here is also stale (as it is on copter as well).

WickedShell, 2018-04-11:
The options for FS_SHORT_ACTN and FS_LONG_ACTN are both missing entries. New ones can be found http://ardupilot.org/plane/docs/parameters.html#fs-long-actn-long-failsafe-action and http://ardupilot.org/plane/docs/parameters.html#fs-short-actn-short-failsafe-action 

The battery failsafes are per battery, with critical/low levels and actions. Documentation for this needs to be generated for Copter as well. The descriptions on BATT_LOW_VOLT BATT_LOW_MAH BATT_FS_LOW_ACT and BATT_LOW_TIMER should give you an idea. (These are all duplicated with CRT instead of LOW for critical failsafes, except LOW_TIMER, that is shared between low and critical). There are also BATT2_ variants for all of these.

Hwurzburg, 2020-03-16:
@WickedShell I think these have been updated now, no?