ArduPilot/ardupilot_wiki #1196: Visually following

ghdmmbyd, 2018-04-13:
Does everyone have an open-source visual follow-up plan, and when does the ArduPilot add visual follow-up?

rmackay9, 2018-04-13:
This isn't really a wiki item so I think I will close this item.

I'm not sure there really is a robust **open source** visual-follow-me feature available.  FlytOS is closed source I think.  I really question whether the dronekit visual-followme works, I don't think I've seen it in action.

As @auturgy suggests, ArduPilot can certainly do the flying part, it is best that it's left to a companion computer to do the vision processing stuff so perhaps that could go into APSync.

closed 2018-04-13 by rmackay9

ghdmmbyd, 2018-04-13:
Thinks！ The visual module is definitely using a coprocessor. What I want is a goal-based follow-up control strategy.

rmackay9, 2018-04-13:
i don't fully understand what "goal based follow-up control strategy" means but perhaps the red-balloon-finder is a decent example of something similar to what you're looking for.  https://github.com/rmackay9/ardupilot-balloon-finder