ArduPilot/ardupilot_wiki #1198: Build and install Ignition Maths for Gazebo

mohand150, 2018-04-13:
hello 
when I execute this code [http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html#compiling-and-installing-gazebo-from-source](url)
it shows me these errors
[image]
thaks for any help !

auturgy, 2018-04-14:
I'm not sure there is an issue with the wiki specifically here, as you're using 14.04, not 16.04 (which the instructions were created to support).
Try swiftgust's build (which also assumes Ubuntu 16.04), or check the Gazebo wiki to see if the dependencies and packages are available for 14.04. Sorry!
If you come up with a way to make things work with 14.04, feel free to make a PR to share what you leanr.

khancyr, 2019-01-24:
instructions have been updated and cleaned.

closed 2019-01-24 by khancyr