ArduPilot/ardupilot_wiki #1200: installing Gazebo From Source

mohand150, 2018-04-13:
Hello,
I tried Gazebo Installation on my new laptop with Ubuntu 14.04 LTS
followed as document here : [(http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html#using-gazebo-simulator-with-sitl)]
it show me this error 
[image]

thaks for any help !

auturgy, 2018-04-14:
The wiki instructions are for Ubuntu 16.04.
This is related to https://github.com/ArduPilot/ardupilot_wiki/issues/1198 so closing to avoid duplication.

closed 2018-04-14 by auturgy

peterbarker, 2018-04-14:
On Sat, 14 Apr 2018, auturgy wrote:

> The wiki instructions are for Ubuntu 16.04.
> This is related to #1198 so closing to avoid duplication.

With @khancyr's help I've got a couple of branches which provide a Vagrant 
virtual machine for ROS.

I'd put these in the "barely working" state at the moment - but they did 
seem to be working...

This one compiles everything from source - these are basically the 
instructions you've been following:
https://github.com/peterbarker/ardupilot/tree/vagrant-ros

This one does things the easy way - packages are provided for modern 
Ubuntus:

https://github.com/peterbarker/ardupilot/tree/recursion-vagrant-ros

They might provide some hints to your problems.

mohand150, 2018-04-15:
@peterbarker excuse but I use the version 16.4.4 TLS
[image]

that's why I can not find the solution to my problem

auturgy, 2018-04-15:
When you opened the issue, you said you were using 14.04.
I see you’ve edited it now.
Please try what peter has suggested above, or try this: https://github.com/SwiftGust/ardupilot_gazebo/blob/master/README.md
which is also intended for 16.04 (happens to work on my machine (Ubuntu 17.10) to).