ArduPilot/ardupilot_wiki #1204: Plane elevon documentation subject to guessing

magicrub, 2018-04-16:
(originally posted by @minid33 from https://github.com/ArduPilot/ardupilot/issues/8179)

I've checked that RC Inputs match the RC Inputs documentation, using a TBS Crossfire via Telemetry Port via a TBS Tango set to APM Mode. Pitch is inverted in RC Setup for plane.

When a user reads the documentation they can complete the servo reversal section and then arrive at a correct setup in MANUAL mode (in more than 1 way, if you include RC setup). In my case this is left servo in channel 1, right servo channel 2, right servo inverted, motor channel 3. These were all set in mission planner's servo setup tab.

When the user gets to the stabilization check portion of the documentation there were previously docs when using the ELEVON_MIXING param for "do X if you see aileron corrections for pitch movements in FBWA mode" [or vice versa].

Now the docs suggest "MANUAL mode assumes your RC inputs have been correctly setup." I'm 100% I do not, because I can see it in the RC setup page of mission planner. Then it suggests "If your movements are incorrect then you need to adjust which servo output is left/right and the reversals of each elevon". There are 8 possible combinations to fix this and it takes a lot of time to try all combinations.

It is also irrational because the left servo setting should always be set to left servo in the channel you placed the left servo connector on the board, same for the right servo. This should mean you have a choice of 4 reversal settings.

Is the user expected to set the elevon mixing to be correct for the stabilization and reverse engineer the RC input?

I would recommend recreating docs that state "when the elevon does X, change Y" for more verbosity.

Version
APM 3.8.4

magicrub, 2018-04-16:
Assigned to @Naterater

auturgy, 2018-04-29:
Closed https://github.com/ArduPilot/ardupilot_wiki/pull/1206

closed 2018-04-29 by auturgy

Naterater, 2019-12-26:
Following up on this, I haven't seen as many double-reversals lately, and I think it's been working well!