ArduPilot/ardupilot_wiki #1209: Plane: add video of how to do a manual launch

rmackay9, 2018-04-18:
I think we should add an instructional video (or two!) showing how to do hand launches.  These videos could go on this page: http://ardupilot.org/plane/docs/automatic-takeoff.html#hand-launching

Most of the advice is written on the page but some things people have said to me that I think are not written:

- throw the vehicle mostly flat, i.e. don't point its nose pointing too high
- don't throw it like a football because that leads to rotation at takeoff
- easiest way is to use Auto mode with a takeoff and RTL command only

Naterater, 2018-04-24:
I would be interested in helping with this.

I'm nearly certain that a takeoff command alone will automatically execute a RTL after.

magicrub, 2018-04-24:
Kavin Garland has a nice launch right [here](https://youtu.be/Yx1k8VgpHlU?t=4). He'd probably make a take-off only video if asked.

rmackay9, 2018-04-25:
@Naterater,
This issue was made for you!  It would be really great if you could help us with the videos.  No pressure but it would be greatly appreciated by many I think.

fnoop, 2018-04-28:
This would be great, especially if it is written for newbies like myself. I've been trying to setup arduplane and get it flying for 6 months (well OK bad weather means rebuilding and the next attempt every couple of months) with no luck so far. 

It's taken a long, long time to read through all the docs and get plane setup, and then work out the easiest way for newbies to get flying is a hand launch and this requires an auto mission to be setup.  Ideally there would be an easier way to hand launch, but at least if the docs talk us through this process it would really help.

So far, ground = 4, flying = 0 :(

Hwurzburg, 2019-12-23:
#2357 awaiting review and merge

closed 2020-01-15 by Hwurzburg