ArduPilot/ardupilot_wiki #1287: Plane - RTL behavior

JeyPi, 2018-06-17:
My today's FPV experience:
I took off from a  spot 280 meters AMSL (altitude above mean sea level)
I was remotely landing the plane in stabilized mode at an RC field that was a 1000 meters away from me and bit lower - 240 AMSL.
I was gliding in for landing. All was great. Just before touch down, the plane lost RC link and the RTL mode kicked in. I was no longer in control. The ardupilot software took a right turn towards the Take off position without any climb. Just a turn. That of course was followed by a crush as the plane was so long and the tip of the wing touched the ground. My RTL altitude was set to 100 meters.
I suggest the code should be adjusted to climb the plane to a safe altitude before turning to the home position. I understand this could not suite to all platforms so at least this behavior could be user pick able.
Video of the event here:
https://drive.google.com/open?id=1M_ZMg-GumfTnKhr5usVDDcZS5AaffCR2

magicrub, 2018-06-17:
This is the behavior that copter does. I've never heard of it being a problem with plane before but I can understand this use case.

magicrub, 2018-06-17:
I just noticed you posted this on the wiki. This is not a documentation thing, you've posted a feature request. Please repost in github.com/ArduPilot/ArduPilot

closed 2018-06-17 by magicrub