ArduPilot/ardupilot_wiki #1320: HeliQuads (Variable Pitch Multicopters)

Califboy1986, 2018-07-11:
[HeliQuads (Variable Pitch Multicopters)](http://ardupilot.org/copter/docs/heliquads.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Califboy1986, 2018-07-11:
Had the sky viper gps drone and it did a flyway on me in full gps mode and to this day have not found it and the company will not do anything about it

rmackay9, 2018-07-12:
This is really a support issue so it should go into the [ardupilot skyviper forum](https://discuss.ardupilot.org/c/arducopter/skyviper) I think although realistically, without a log file it would be hard to get to the bottom of it.

Sorry for your loss :-(.

If it's any consolation it looks like the drones are now on sale for only $53 at target.

closed 2018-07-12 by rmackay9