ArduPilot/ardupilot_wiki #1322: Radio Control Calibration in Mission Planner; no green bars visible

KonradPW, 2018-07-12:
[Radio Control Calibration in Mission Planner](http://ardupilot.org/plane/docs/common-radio-control-calibration.html)
I am trying to do the RC Calibration in Mission Planner.
I have a Taranis X9D+ and a R-XSR bound.
The R-XSR is connected via the Craft and Theory Telemetry Cable to the PixHawk.
My Channels are 1=Throttle, 2=Aileron, 3=Elevator, 4=Rudder.
I am using a SiK Telemetry Radio for the connection to the PC. The connection is stable.
Before and during the calibration there are no moving green bars present, "axes" are empty so to say.
Literally nothing happens, no matter what I do on the Taranis.
When ending the calibration I get the message "Bad channel 1 input, canceling".

sslaughter, 2018-07-21:
Konrad,

Usually the green bars don't show up if it's not getting anything from the receiver, could you confirm the following:
- You have the servo portion of the cable plugged into RCIN on the Pixhawk
- In Mission Planner, in the Config/Tuning tab, "Planner" section, under telemetry rates, the "RC" drop-down is >0
- For the serial port you have the telemetry cable plugged into, the SRX_RC_CHAN parameter is > 0.

Finally, I don't really know, but this question may be more appropriate for the discuss forums rather than as a wiki issue.

milad-akbari-github, 2019-08-07:
Hi everyone !!! i have this problem too !!! i use radiolink T8FB and R8EF reciver !!! and Pixhawk !! 
i check reciver with one servomotor and it work with PPM & S-BUS !!! but on mission planner green bar not apear !!

rmackay9, 2019-08-07:
This is not the place for support though I'm afraid.  This is the to-do list for the wiki so I will close this issue.  A much better place to get help is in the [Copter forums](https://discuss.ardupilot.org/c/arducopter/copter-36) or https://discuss.ardupilot.org/c/ardurover/rover-35.

closed 2019-08-07 by rmackay9