ArduPilot/ardupilot_wiki #1357: Git Submodules

tianzhongjin, 2018-08-03:
I have a problem like below.
I am quite newbie.
I just follow this site 
http://ardupilot.org/dev/docs/git-submodules.html
But it didn't help me.
Please help me what is going on?

message i got is below

$ git submodule update --recursive
fatal: Not a git repository: ../../i:/Drone/ardupilot/.git/modules/modules/PX4Firmware/modules/Tools/gencpp
Unable to find current revision in submodule path 'Tools/gencpp'
fatal: Not a git repository: ../i:/Drone/ardupilot/.git/modules/modules/uavcan/modules/dsdl
Unable to find current revision in submodule path 'dsdl'
Failed to recurse into submodule path 'modules/PX4Firmware'
Failed to recurse into submodule path 'modules/uavcan'

auturgy, 2018-08-03:
Did you fork and clone the repo first?
Confirm that you followed this: http://ardupilot.org/dev/docs/git-fork.html
then this: http://ardupilot.org/dev/docs/git-clone.html#git-clone
Then from within the ArduPilot/ardupilot directory, type git submodule update —init —recursive 

Regards,

James

tianzhongjin, 2018-08-04:
Thank you for replying my question.
I just found the solution.
In failed case, I  used PX4Console which is included in PX4 Toolchain.
So, I simply change to use cygwin64 console and then try it the same.
I worked!!
I don't know why PX4Console doesn't work, I'm not an expert anyway.

Well, now anyway i can start building source files.
And building was no problem so far.

Thank you !!

closed 2018-08-05 by tianzhongjin