ArduPilot/ardupilot_wiki #1365: typo in RC_OPTIONS description

Kelly-Foster, 2018-08-10:
typo in "Ignore RC Reciever"
i before e, except after c

khancyr, 2018-08-10:
hello,
could you point out in which page or the code ? 
Or, if you want you should have a button :  Edit on GitHub, that will allow you to submit directly a modification !

WickedShell, 2018-08-10:
@khancyr It's param docs, soo it can't be edited on the wiki. I'm afraid I have to fess up to this being my bad spelling (again).

khancyr, 2018-08-10:
that what I feared ... 
That is here : https://github.com/ArduPilot/ardupilot/blob/master/libraries/RC_Channel/RC_Channels_VarInfo.h#L89
@Kelly-Foster want to correct it ?

hamishwillee, 2018-08-10:
> i before e, except after c

Apparently they've stopped teaching that in schools in Victoria, Australia now. Too many exceptions!

auturgy, 2018-08-10:
Unacceptable!

khancyr, 2019-01-24:
close with https://github.com/ArduPilot/ardupilot/pull/10334

closed 2019-01-24 by khancyr