ArduPilot/ardupilot_wiki #1375: Got Bad Lidar Health and couldn't find RNGFND_LANDING

cyc607, 2018-08-16:
[Rangefinders (landing page)](http://ardupilot.org/plane/docs/common-rangefinder-landingpage.html) -

I use ArduRover v3.4.2 and pixhawk with RPLidar A2.
When I connect  5v and GND to telem1 the voltage decrease from 5v to 0v.
So I only connect RX TX on pixhawk and 5v GND MotorCTL is from another power supply.
I always get bad lidar health on mission planner and can't find RNGFND_LANDING parameter.

vierfuffzig, 2019-06-03:
RNGFND_LANDING parameter is plane specific, so is the respective note on the rangefinder landing page. you only see it if you access that page from arduplane wiki.

rmackay9, 2019-06-04:
This definitely looks like a support issue so this should go to the [rover forums](https://discuss.ardupilot.org/c/ardurover).  I think I will close this issue.

closed 2019-06-04 by rmackay9