ArduPilot/ardupilot_wiki #1412: Setting up SITL using Vagrant Broken

hamishwillee, 2018-09-11:
Followed instructions in [Setting up SITL using Vagrant](http://ardupilot.org/dev/docs/setting-up-sitl-using-vagrant.html). Do not work.

Instructions suggest using rsync. I installed this via cygwin. BUT if you try to uncomment the sharedfolders line `# config.vm.synced_folder ".", "/vagrant", type: "rsync", rsync__auto: true` you note that this is not present any more. The whole vagrant structure appears to have changed and support multiple base OS and init files. 

1. Do you need to install rsync on host still? If so, how is this "enabled"
2. Can I have a brief update on how the new scripts work - ie how do you choose different bases.

Anyway, at end of installation instruction, tried to run SIM. There is some issue related to Python. Thoughts?
```
D:\github\forks\ardu\ardupilot>vagrant ssh -c "sim_vehicle.py -j 2"
Ardupilot environment ready.  Run 'sim_vehicle.py' to start simulating an arducopter instance.
To build for fmuv2:
  cd /vagrant
  ./waf configure --board=px4-v2
  ./waf build --target=bin/arducopter
/usr/bin/env: ‘python\r’: No such file or directory
Connection to 127.0.0.1 closed.
```

3. Thoughts on debugging this?

peterbarker, 2018-09-11:
On Mon, 10 Sep 2018, Hamish Willee wrote:

> Followed instructions in Setting up SITL using Vagrant. Do not work.
>
> Instructions suggest using rsync. I installed this via cygwin. BUT if you try to uncomment the sharedfolders line # [...]

I suggest we nuke those instructions unless someone knows how to get that 
working.

>  1. Do you need to install rsync on host still? If so, how is this "enabled"
>  2. Can I have a brief update on how the new scripts work - ie how do you choose different bases.

vagrant up xenial32
vagrant ssh xenial32

> Anyway, at end of installation instruction, tried to run SIM. There is some issue related to Python. Thoughts?
>
> D:\github\forks\ardu\ardupilot>vagrant ssh -c "sim_vehicle.py -j 2"
> Ardupilot [...]

That definitely looks like a cr/lf issue.  Did any of those waf commands 
run to completion?  If so, tracking down which script has that '\r' in it 
would be useful.

peterbarker, 2018-09-17:
On Sun, 16 Sep 2018, Hamish Willee wrote:

> The script that is failing is sim_vehicle.py itself (debugged simply by changing the first line from #!/usr/bin/env python to #!/usr/bin/env python2).

So that does make me curious - if you change that to "#!/usr/bin/env 
python3" does the problem reappear?

> I didn't do anything explicitly to change the line endings at any point, though I guess it is possible that git did. Shouldn't the system be resistant to
> that? / What can I do?

Yes, git has a setting for this.

https://git-scm.com/book/en/v2/Customizing-Git-Git-Configuration (search 
for autocrlf

Also search for autocrlf here:

http://ardupilot.org/dev/docs/git-install.html

hamishwillee, 2018-09-17:
Thanks @peterbarker . I used dos2unix in the VM and worked. Worth adding a note about the error and possible solution to the doc itself? I suspect that people might find this doc by searching.

I created #1425 to remove the rsync stuff. Building for the simulator wasn't lightning fast, but wasn't prohibitive either. 
Who would know whether this is still needed?

hamishwillee, 2018-09-18:
@peterbarker Thanks! I'll close this. I'd rather understand how rsync should work, but pragmatically we should merge #1425 because it is more important that users don't get confused.

closed 2018-09-18 by hamishwillee