ArduPilot/ardupilot_wiki #1419: OpenPilot Revolution and RevoMini

MFlghts, 2018-09-16:
[OpenPilot Revolution and RevoMini](http://ardupilot.org/plane/docs/common-openpilot-revo-mini.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

**Optional Settings**

- I2C on Flexi port
- RevoMini’s OPLink port can be used for external SD card connection
- arbitrary pins can be used for various connections like 2nd GPS, additional PWM outputs etc. within certain restrictions

Please describe how exactly to do all these setups!

iforce2d, 2019-01-19:
This might be helpful regarding the I2C on flexiport:
http://www.iforce2d.net/tmp/revomini-pin-change-i2c

vierfuffzig, 2019-06-02:
as noted on the referenced hardware page, these changes require fundamental changes to the board's hwdef and compiling a custom firmware. a comprehensive howto on customizing firmware well exceeds the purpose of the default hardware page though. see http://ardupilot.org/dev/docs/porting.html for further detail.

closed 2019-09-11 by Hwurzburg