ArduPilot/ardupilot_wiki #1420: Replace Tarot as recommended Gimbal

hamishwillee, 2018-09-17:
The 3DR Tarot Gimbal is no longer manufactured (it is still available, but this will be temporary). I have therefore archived the article in #1409) and cleaned up links.

The Tarot was previously the "recommended" gimbal. This issue is a placeholder requesting clarification of what is recommended  now (and why) and also for information on other gimbals for which documentation might be useful.

auturgy, 2018-09-17:
@jmachucca77
This might be a candidate for code-in?

Regards,

James

rmackay9, 2018-09-17:
the tarot gimbal is still available though.  I'm not sure it's really the recommended gimbal anymore but it is still available and it's pretty reliable.

hamishwillee, 2018-09-17:
It is available in theory, but only from Amazon. As soon as those stocks go. Gone. I archived it as suggested here: https://github.com/ArduPilot/ardupilot_wiki/pull/1409#issuecomment-420122661

We can undo if you strongly disagree - I did wait 6 days for further comment.

jmachuca77, 2018-09-17:
Writing the wiki page might be a candidate for code-in... but the problem is what hardware should it describe, plus not sure these students will have a budget to buy a gimbal and necessary hardware to actually test it.

rmackay9, 2019-06-02:
I think we can close this now..

closed 2019-06-02 by rmackay9