ArduPilot/ardupilot_wiki #1423: Four Channel Planes

daskalovst, 2018-09-17:
[Four Channel Planes](http://ardupilot.org/plane/docs/guide-four-channel-plane.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

On the RC input section have mistake rudder and elevator reaction is wrong.

auturgy, 2018-09-19:
The RC Input section is testing responses to RC stick inputs, not stabilisation reactions.  The expected responses to RC Input for rudder (yaw) and elevator (pitch) are correct, as the input is trying to cause a resonse in the given direction.  If your aircraft is responding in the opposite direction, you need to reverse them.
Testing stabilisation should give the opposite response, as the controller is attempting to fight the movement, not cause it.

auturgy, 2018-09-19:
https://github.com/ArduPilot/ardupilot_wiki/pull/1427
Rudder left was wrong.  Thanks. (and thanks @WickedShell)

closed 2018-09-19 by auturgy