ArduPilot/ardupilot_wiki #1454: Connection List Problem

DIYalways, 2018-10-23:
I tried to connect multiple vehicle in mission planner, and succeed in connecting one by one.
I find there is a connection list option, but when I chose this one, It ask me to upload a file.
I don't know the proper format.

I want to figure out two things. 
1. How to save the connection(Manuel connected)?
2.How to upload a connection list（what format and what type？）

lucasantoro, 2019-04-16:
any idea how to use this connection list? in order to connect at once multiple uavs?

rmackay9, 2019-04-16:
These are really support questions or enhancement requests for Mission Planner I think so I might close this.  More likely to get an answer on the [MP issues list](https://github.com/ArduPilot/MissionPlanner/issues) or in the forums I think.

closed 2019-04-16 by rmackay9