ArduPilot/ardupilot_wiki #1477: What is the meaning of different branchs on APM source code repertory?

ejngnng, 2018-11-15:
Hi all, 
I confuse about the ardupilot source code branchs，especially about ArduCopter-3.2.x and Copter-3.x，they are all about copters, what's the difference? anyone can help, thanks!

hamishwillee, 2018-11-15:
Every time a new release is made off the master codeline a new branch is created. That allows both the new release and the main code line to evolve separately (though changes to the branches only happen for a bit, until critical errors and required features are present).

The branches are all for "Copter". The name change is just because the frame type is now referred to as Copter in most places rather than ArduCopter. Later numbers reflect later releases.

closed 2018-11-15 by hamishwillee

ejngnng, 2018-11-15:
> Every time a new release is made off the master codeline a new branch is created. That allows both the new release and the main code line to evolve separately (though changes to the branches only [...]

Thanks a lot  :)