ArduPilot/ardupilot_wiki #1516: Translation support

bys1123, 2018-12-06:
Hi, dear ardupilot team

I'm the guy that made APM chinese site (ardupilot.cn) years ago, led a developer team translated almost 200 docs, and translated Mission Planner.

But I left drone industry for a while and now I'm back. I helped PX4 team build their localization platform first. it‘s on Crowdin, free for open source project, really help for translators cooperate with each other.

Is this project ready for translation? Can you build a crowdin project for translators, I think this platform support reStructuredText too. I can lead a team join in this on chinese translation.

auturgy, 2018-12-06:
Hi Stone!
Great to hear from you!
I remember when you led the efforts to provide a Chinese language version of Missionplanner.
We would certainly support and appreciate your assistance with making our wiki available to more people.
@proficnc can you suggest a good contact?

proficnc, 2018-12-06:
@bugobliterator, who in our China office would be a good contact? Maybe Feels?

bys1123, 2018-12-06:
@proficnc @auturgy Thanks for your reply. Glad to talk with Ardupilot china office. 
And I think this need @rmackay9 create a project on Crowdin as a project manager. 
https://crowdin.com/page/open-source-project-setup-request

auturgy, 2018-12-06:
Just so you know, I’m the guy listed here: http://spi-inc.org/projects/ardupilot/ so if it’s ok with you I’m more than happy to kick things off.
But it’s definitely fine for @rmackay9 to kick it off too.

hamishwillee, 2018-12-06:
You need to do a little more testing/thinking before jumping into this. ArduPilot wiki is a different infrastructure to PX4 wiki, and may require quite different approaches.

PX4 uses gitbook with markdown files. Crowdin pulls in the markdown pages from github, tracks changes in Github to find new strings, and pushes out translated pages to the correct locations for the internationalisation mechanism. The thing is, Crowdin knows markdown, and can handle this properly. It does not know reStructured text, and this may break things - so it needs a little testing.

Secondly, if you use this approach you're committing to a completely separate build for each language and URL. That is fine, but you need to be aware of it.

Lastly, Sphinx inbuilt language support is via POTs files. Crowdin does support those, but it won't be such a good experience for the translators because there is less context. 

Anyway, I'd be tempted to take the POTS approach, but a little testing required of the options.

auturgy, 2018-12-06:
No worries. I’ve opened a request

bys1123, 2018-12-06:
@hamishwillee Thanks for your suggestion, very helpful. I mixed up Crowdin support format list.
I think POT files is not a good option, it will be nightmare if no context. I'll figure out is that support handle .rst as .txt.

bys1123, 2018-12-06:
[image]
[image]
I think translate as .txt will work well, but I can't create free project, can't continue.

auturgy, 2018-12-06:
Give us a couple of days to work through how to best facilitate this: there are a few questions we need to ask ourselves :)

bys1123, 2018-12-06:
Wow~ Crowdin approved my application, you can take a look how it works.
https://crowdin.com/project/ardupilot
This project is based on my forked repo, hard to sync with official repo. I will wait the official translation project.
This doc would helpful on Integration: https://support.crowdin.com/vcs-integrations-configuring-online/

bys1123, 2019-03-05:
Any new updates? @auturgy

Hwurzburg, 2026-09-26:
I am closing this...its very old, and AI (ChatGPT) offers fairly robust translation capabilities now ....

closed 2026-09-26 by Hwurzburg