ArduPilot/ardupilot_wiki #1531: Full parameter tree - too long

Naterater, 2018-12-22:
OSDn_, BATTn_, RCn_, and SERVOn_ are all parameters that have multiple instances and really only require one description per for all instances. The tree is getting out of hand with its length, and the OSD parameters have at least doubled the size of the tree.

Is there any way to automatically shrink down this menu or condense the repetitive items?
[image]

vierfuffzig, 2018-12-22:
+1
collapsing OSD params would add massive convenience

khancyr, 2018-12-22:
We could add something on the parser to shrink the osd parameters

vierfuffzig, 2018-12-23:
is it /Tools/autotest/param_metadata/htmlemit.py that generates the param page? this scripting is quite a bit beyond my scope, can't figure how exactly the toc depth is generated. seems the parser's depth is limited to 1 while the OSDn params have 2 subgroups actually...

Hwurzburg, 2020-01-15:
this seems to be fixed...OSD parameters are collapsed unless expanded...

closed 2020-01-15 by Hwurzburg